lyusupov/SoftRF #171: Devices using SoftRF are not visible to GXAirCom or Skytraxx using any protocol

cyberorg, 2024-09-22:
**ATTENTION** ! 

- Use this form to report bugs in an [**Official Release**](https://github.com/lyusupov/SoftRF/releases) of SoftRF firmware, hardware and web pages only ;
- Issue reports from anonymous users will be most likely invalidated. Reasons for that are well explained [here](https://berthub.eu/articles/posts/anonymous-help/) ;
- Prior to asking a question, look into ["Closed issues"](https://github.com/lyusupov/SoftRF/issues?q=is%3Aissue+is%3Aclosed) area to see if it was answered before ;
- No any "ideas" or new feature requests, please ! New feature requests do not subject for taking into consideration unless they come from co-developers ;
- "Enhancements" of existing features are only allowed when you are ready to implement them by yourself. See [CONTRIBUTING](https://github.com/lyusupov/SoftRF/blob/master/CONTRIBUTING.md) list for details ;

Recommended:
- screenshot or photo width - from 600 to 800 pixels ;
- aspect ratio - 4:3, 3:4, 16:9 or 9:16.
- substitute 'space' by 'x' character to mark a hardware checkbox

--------------------

## Hardware

- [ ]   Badge

< photo of your hardware > (**required**)
[image]

## Firmware version
Multiple
<  screenshot of WebUI and/or [**SoftRF Tool**](https://github.com/lyusupov/SoftRF/wiki/SoftRF-Configuration-Tool) status page > (**required**)

## Firmware settings

<  screenshot of WebUI and/or [**SoftRF Tool**](https://github.com/lyusupov/SoftRF/wiki/SoftRF-Configuration-Tool) settings page > (**required**)
[image]

I have tried different protocols as well as region frequencies.

## Describe the bug
Devices using SoftRF are not visible to GXAirCom or Skytraxx using any protocol. GXAirCom ground station can see Skytraxx device on FANET and vice-versa, but neither of them see SoftRF device. All the SoftRF devices can see each other.

## To Reproduce
Power on SoftRF device near Skytraxx or GXAirCom.

## Expected behavior
The devices should be visible to each other using FANET protocol.

Also reported here: https://github.com/gereic/GXAirCom/issues/169

lyusupov, 2024-09-22:
Works fine to me with GXAirCom

[image]

<br>
<br>

**Resolution**: unable to reproduce

closed 2024-09-22 by lyusupov

cyberorg, 2024-09-22:
This is the config I am using, could you share yours?

$PSRFC,1,0,5,1,7,1,0,2,2,1,0,1,1,5,0,0,0,0,0*4F

I downgraded the GXAirCom to the lowest available at the moment to test also, didn't find the version you are using. I was using the latest version.

Here are the 3 devices, as you can see GXAirCom and skytraxx see each other, but neither show SoftRF and the t-echo also show 0 ACFTS.

I have tried on multiple t-echos, they can see each other.

[image]

lyusupov, 2024-09-22:
Ask SoftRF users Gitter community https://gitter.im/lyusupov/SoftRF to help you.