lyusupov/SoftRF #175: No aircraft data on Badge (workaround known)

jorisrobijn, 2024-10-26:
## Hardware

- [ ]   Standalone
- [x]   Badge
- [ ]   Prime Mark III
- [ ]   Prime Mark II
- [ ]   Dongle
- [ ]   SkyView EZ
- [ ]   SkyView Pico
- [ ]   Uni
- [ ]   Ham
- [ ]   Midi
- [ ]   Mini

- [ ]   Academy
- [ ]   Lego
- [ ]   ES Edition
- [ ]   Balkan
- [ ]   Raspberry Edition
- [ ]   UAV
- [ ]   WebTop Serial
- [ ]   WebTop USB
- [ ]   UAT978 module
- [ ]   Flight Recorder

[image]

## Firmware version

[image]

## Firmware settings

[image]
[image]

## Describe the bug

The aircraft database from OGN is not loaded while it is present.

## To Reproduce

Use 1.5.1 firmware and start the device. Install the OGN aircraft database as indicated on page https://github.com/lyusupov/SoftRF/wiki/Badge-Edition.-Aircrafts-database

## Expected behavior

Aircraft database is loaded.

## Workaround and probable solution

As documented in issue #156 , the problem appears to be with the type of file. It is not set to be an OGN type of file, it appears. 

To fix, connect over USB to the device via virtual serial port and send the string $PSKVC,1,0,0,0,2,1,0,,,0,0,3,3,0,0,0,0,0,000000*73 immediately followed by an Enter. Or as done in #156, use the bluetooth terminal to sent the same. This sets the file type to OGN (together with other minor settings which cannot be prevented this way) .

After doing this, without re-putting the OGN database on the device again, the database was loaded.

I think it's best to implemented the solution mentioned in #156 .

lyusupov, 2024-10-26:
The feature works fine to me.

This is photo taken with **up to date aircrafts database** (Oct 26 of 2024) loaded into the SoftRF **Badge** with firmware [Release 1.5.1](https://github.com/lyusupov/SoftRF/releases/tag/1.5.1):

[image]

<br>
<br>

**Resolution**: unable to reproduce

closed 2024-10-26 by lyusupov

jorisrobijn, 2024-10-26:
That is how I have it now as well. However, it did not work without setting the setting in the very difficult way. The setting is not available in the Android app. I guess you have set the setting (maybe long ago) in some other way.
Is it maybe possible to just try to load both database types? Then you don't need the setting anymore.

lyusupov, 2024-11-08:
Wrong value of aircraft database selector is supplied by **SoftRF Tool**:

[image]

[image]

<br>
<br>

The [**SoftRF Tool**](https://play.google.com/store/apps/details?id=com.dda.softrftool) is a third-party app which has nothing to do with this SoftRF GitHub repo. Ask for support (if any) on the [app's home page](https://play.google.com/store/apps/details?id=com.dda.softrftool) in Google Play.

Read the [**Disclaimer**](https://github.com/lyusupov/SoftRF/wiki/SoftRF-Configuration-Tool#disclaimer) 
[image]

<br>
<br>

**Resolution**: invalid