lyusupov/SoftRF #177: GDL90 option not available in SoftRF Tool

d3mac123, 2024-12-24:
## Hardware: T-Echo 915Mhz
## Firmware version: s140 6.1.1 - Software revision: 1.3
[image]

[image]

## Firmware settings

<  screenshot of WebUI and/or [**SoftRF Tool**](https://github.com/lyusupov/SoftRF/wiki/SoftRF-Configuration-Tool) settings page > (**required**)

## Describe the bug
I just got a new phone (purchased only to configure the T-Echo), and installed the APK (SoftRF Tool is not available in the Google Play store) but I cannot see the GDL90 option. As I am in the US, I only need to see ADSB traffic. How do I make the option show up in the SoftRF Tool app?

## To Reproduce

## Expected behavior

d3mac123, 2024-12-24:
After posting it, I installed firmware 1.6 but got the same results: no GDL90 option.

lyusupov, 2024-12-25:
* ADS-B is not listed in the **Badge Edition** [protocols specs](https://github.com/lyusupov/SoftRF/wiki/Badge-Edition#compatibility-3) ;
* the [**SoftRF Tool**](https://play.google.com/store/apps/details?id=com.dda.softrftool) is a third-party app which has nothing to do with this SoftRF GitHub repo. Ask for support (if any) on the [app's home page](https://play.google.com/store/apps/details?id=com.dda.softrftool) in Google Play ;
* read the [**Disclaimer**](https://github.com/lyusupov/SoftRF/wiki/SoftRF-Configuration-Tool#disclaimer) 
[image]

**Resolution:** invalid

closed 2024-12-25 by lyusupov