lyusupov/SoftRF #185: Standalone version bootloop in "Legacy protocol"

legacy60000, 2025-06-27:
**ATTENTION** ! 

- Use this form to report bugs in an [**Official Release**](https://github.com/lyusupov/SoftRF/releases) of SoftRF firmware, hardware and web pages only ;
- Issue reports from anonymous users will be most likely invalidated. Reasons for that are well explained [here](https://berthub.eu/articles/posts/anonymous-help/) ;
- Prior to asking a question, look into ["Closed issues"](https://github.com/lyusupov/SoftRF/issues?q=is%3Aissue+is%3Aclosed) area to see if it was answered before ;
- No any "ideas" or new feature requests, please ! New feature requests do not subject for taking into consideration unless they come from co-developers ;
- "Enhancements" of existing features are only allowed when you are ready to implement them by yourself. See [CONTRIBUTING](https://github.com/lyusupov/SoftRF/blob/master/CONTRIBUTING.md) list for details ;

Recommended:
- screenshot or photo width - from 600 to 800 pixels ;
- aspect ratio - 4:3, 3:4, 16:9 or 9:16.
- substitute 'space' by 'x' character to mark a hardware checkbox

--------------------

## Hardware

- [x]   Standalone
- [ ]   Badge
- [ ]   Prime Mark III
- [ ]   Prime Mark II
- [ ]   Dongle
- [ ]   SkyView EZ
- [ ]   SkyView Pico
- [ ]   Uni
- [ ]   Ham
- [ ]   Card
- [ ]   Handheld
- [ ]   Nano
- [ ]   Cozy
- [ ]   Midi
- [ ]   Mini

- [ ]   Academy
- [ ]   Lego
- [ ]   ES Edition
- [ ]   Balkan
- [ ]   Raspberry Edition
- [ ]   UAV
- [ ]   WebTop Serial
- [ ]   WebTop USB
- [ ]   UAT978 module
- [ ]   Flight Recorder

< photo of your hardware > (**required**)

## Firmware version

<  screenshot of WebUI and/or [**SoftRF Tool**](https://github.com/lyusupov/SoftRF/wiki/SoftRF-Configuration-Tool) status page > (**required**)

## Firmware settings

<  screenshot of WebUI and/or [**SoftRF Tool**](https://github.com/lyusupov/SoftRF/wiki/SoftRF-Configuration-Tool) settings page > (**required**)

## Describe the bug
MCU ESP8266 Bootloop with legacy protocol on the moment of GPS FIX. Until GPS FIX no issue but on moment of the lock it goes to bootloop until there is good GPS signal. On moment the GPS have no lock the bootloop stops. With OGTP there is not issue. If I use MCU the AI Thinker one it works fine no bootloop. Only witj ESP8266 with GOS lock.

## To Reproduce
Set the device in Legacy and wait for GPS Lock MCU must be ESP8266

## Expected behavior

[image]
[image]
[image]

lyusupov, 2025-06-27:
* reported version is *Release 1.3** which is not up to date. Try recent [**Release 1.7**](https://github.com/lyusupov/SoftRF/releases/tag/1.7) then report if it remains there
* possible duplicate of  https://github.com/lyusupov/SoftRF/issues/13
* use **ESP32** or other modern MCU with the **Standalone Edition**   ESP8266 makes no sense these days.

**Resolution:** duplicates #13

lyusupov, 2025-07-13:
@legacy60000 

Unzip and try this ESP8266 firmware binary file out:    [**SoftRF.ino.zip**](https://github.com/user-attachments/files/21204357/SoftRF.ino.zip)

<br>
<br>

It works fine to me on the ESP8266 **Standalone Edition** hardware -

[image]

[image]